UM-ARM-Lab/kuka_iiwa_interface #122: Upgrade realtime from 16.04

bsaund, 2020-11-25:
Our code stack is moving towards noetic and away from kinetic.
E.g. https://github.com/UM-ARM-Lab/arc_utilities/pull/105

Upgrading realtime to 18.04 might be relatively painless.
Upgrading to 20.04 would likely be more complicated

PeterMitrano, 2020-11-28:
I just tested and this builds as-is on realtime. Why did we think we needed to upgrade? I still think we _should_ update but it seems like we don't _need_ to, at least not yet.

bsaund, 2020-11-29:
I believe there was a concern if realtime's gcc supported C++17

PeterMitrano, 2020-11-29:
Yea although it seems were not using it at the moment. If that's the only concern it should be super easy to install gcc9 or something. We can built it from source if need be.

PeterMitrano, 2020-11-29:
Is there anything special about realtime? Does it actually run the realtime linux kernel?

dmcconachie, 2020-11-29:
It is using a realtime patched kernel; yes.

PeterMitrano, 2020-11-29:
....why? it's afaik it's not running any code that is realtime essential. It's just the arm wrapper node

dmcconachie, 2020-11-29:
It's something @calderpg did; I had the same vague question, but did not pursue it. I think at one point it was due to using the FRI interface for Victor, which requires a very steady 1 kHz signal.

calderpg, 2020-11-29:
Because in the spring of 2017, it looked like FRI in torque mode was going to be the best way to achieve compliant motions (it wasn't, for various reasons). Even now, FRI is best used on realtime-patched kernels, although if you have to, core reservation+pinning can be used with some success.

Anyways, PREEMPT-RT patches are easy enough to apply for 18.04 and 20.04, so it shouldn't stop you from updating.

PeterMitrano, 2020-11-29:
Thanks! that makes sense, I forgot about FRI. I agree it should be easy to upgrade to a 20.04 realtime kernal. Let's do this after the move & when no one is going to need the robot for a week @powertj

bsaund, 2020-12-29:
I'm going to upgrade realtime.

Our new stack does not currently build on realtime. `robot_3f_gripper_articulated_gazebo_plugins` requires c++20 (or at least 17). Rather than fiddling with gcc updates, I'll just upgrade

bsaund, 2020-12-30:
Realtime is now on 20.04. I did no additional patching to make it a "realtime" version

closed 2020-12-30 by bsaund