UM-ARM-Lab/kuka_iiwa_interface #130: Decide on location of manual motion script

bsaund, 2021-01-12:
Currently `manual_motion` is duplicated in KHI and `arm_robots`. 

`manual_motion` does not require `arm_robots` code, which was more important with the openrave dependency of `arm_or_robots`. With the open_rave dependency dropped, we can migrate everything to `arm_robots` to keep all victor motion scripts in the same place

PeterMitrano, 2021-01-12:
I think I prefer it being in KHI if it doesn't need/use arm_robots

PeterMitrano, 2021-01-12:
but I guess it depends on what you mean by "everything"?

bsaund, 2021-01-12:
I misspoke. I just mean migrating `manual_motion`.

PeterMitrano, 2021-01-12:
I'd say it makes more sense in VHI (I can never keep these names straight), since it's specific to that code

bsaund, 2021-01-12:
The VHI version will need to be fixed to work with the other changes we've made to KHI. I think some namespaces have changed.

PeterMitrano, 2021-01-12:
I think someone has used manual motion recently @powertj @LemonPi @mvandermerwe  which version did you use?

bsaund, 2021-01-12:
Over winter break the `arm_robots` version worked, the KHI one did not.

mvandermerwe, 2021-01-12:
I did not use manual motion recently.

PeterMitrano, 2021-01-13:
in that case I'd say move the arm_robots version to VHI