UM-ARM-Lab/kuka_iiwa_interface #149: Test out https://github.com/ICube-Robotics/iiwa_ros2 for victor

PeterMitrano, 2023-10-24:


PeterMitrano, 2023-11-30:
So it turns out that using the gripper is actually the tricky part here. Using the iiwa_ros2 java file is really simple, I got it working in the sense that the FRI interface is started and waiting for commands. So some next steps are:

- [ ] Finish setting up the C++ FRI ROS2 side of things on the kukacontrol machine
- [ ] Calibrate the gripper since the arm still needs to know the mass of the robotiq gripper even though it's not going to be controlling it directly anymore
- [ ] Get the robotiq working. This means getting some code on the kukacontrol machine that works and making/using the right cables to connect the X651 port to the kuka control machine + power

PeterMitrano, 2023-11-30:
I think a next step is to try to modify the controller to control both arms. Some steps (probably there are more)

- [ ] start by forking the iiwa_ros2 repo into the armlab org
- [ ] make an updated URDF with both arms. Probably start with the new urdf because theirs is very pretty, and just reference our victor.urdf to make a new victor.urdf. For now we can ignore the gripper in the urdf.
- [ ] include all 14 joints (this will require renaming them!) [here](https://github.com/ICube-Robotics/iiwa_ros2/blob/main/iiwa_description/ros2_control/iiwa.r2c_hardware.xacro) and [here](https://github.com/ICube-Robotics/iiwa_ros2/blob/main/iiwa_description/config/iiwa_controllers.yaml)
- [ ] In `IiwaFRIHardwareInterface`, create _two_ `robotClient_` objects, one for left and one for right, and update the code to handle that. We can just assume a fixed ordering (left then right) in the vectors.

bsaund, 2023-11-30:
> However, the bad news is joint2 on the left arm is now making a crunchy bad noise.

FYI this happened before and we sent the robot back to Kuka for servicing. We had a loaner white kuka iiwa for awhile. I don't remember which arm

PeterMitrano, 2024-01-18:
abandoning this in favor of just upgrading the stack in-place

closed 2024-01-18 by PeterMitrano