UM-ARM-Lab/kuka_iiwa_interface #152: shuffling computers on the cart

umicharmlab, 2024-02-16:
I'd like to see if we can fix the following issues:
 - we are using two 10-year-old computers running old versions of windows
 - the ubuntu machine used for the kukas is very slow, which means compiling moveit takes forever
 - we can't run VR on any of these computers which means VR teleop for victor isn't possible

Proposed solution:
- Setup Loki to be the ubuntu control machine
- Setup armistice with windows so it can be used to run VR, Sunrise, and Mocap

TODOS
- [ ] Test using Loki as the control machine
- [ ] Finish removing ubuntu from armistice and try updating to windows 11
- [ ] Try installing Vicon Tracker, Sunrise Workbench, and Steam/Unity on armistice
- [ ] Test everything on armistice

PeterMitrano, 2024-03-14:
Tracker 3 doesn't seem to work on Windows 11, so we can't run mocap on Armistice. We could upgrade to Tracker 4 but that costs like 5k and isn't strictly necessary. So for now I think we should just go back to using the windows 10 computer to run mocap.

PeterMitrano, 2024-03-26:
We are currently working with Vicon to fix the problem. We've tried reverting to the windows 10 version but can't get the cameras to connect.